rwaldron/galileo-io #24: new io.Gpio(this.addr) fails - again, but I found out something interesting

Injac, 2015-01-12:
Hi,

I have to say that I am starter with Edison and Galileo.But here is what I have done:

- Compiled MRAA on my Edison
- Built node.js (latest release) on my Edision
- Installed johnny-five, galileo-io and the alias package edision-io

Every single sample I tried using edision-io, the johnny5 samples did throw the following error:

```javascript
/root/node/node_modules/galileo-io/lib/mraa.js:66
    state.io = new io.Gpio(this.addr);
               ^
Error: Illegal arguments for construction of _exports_Gpio
    at new Pin (/root/node/node_modules/galileo-io/lib/mraa.js:66:16)
    at Galileo.<anonymous> (/root/node/node_modules/galileo-io/lib/galileo.js:172:16)
    at Array.map (native)
    at new Galileo (/root/node/node_modules/galileo-io/lib/galileo.js:168:24)
    at new Board (/root/node/node_modules/johnny-five/lib/board.js:295:17)
    at Object.<anonymous> (/root/node/test3.js:4:9)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)

```

But, if I use the MRAA sample here:

https://github.com/intel-iot-devkit/mraa/blob/master/examples/javascript/Blink-IO.js

It works. And I have no idea why.

Really hope this helps to nail down the issue.

rwaldron, 2015-01-12:
> Compiled MRAA on my Edison

This shouldn't be necessary, Galileo-IO will install the latest version when you do `npm install galileo-io`

-------

1. Run this and tell me what the result is: `node -e 'console.log(require("mraa").getVersion());'`
2. Run: 
```
echo "src mraa-upm http://iotdk.intel.com/repos/1.1/intelgalactic" > /etc/opkg/mraa-upm.conf;
opkg update;
opkg install libmraa0;
```

Are you on the arduino expansion board or the mini board?

Injac, 2015-01-12:
Here is the output:

v0.5.4-66-g62c3264

I have done that already. Added the intergalactic a few times. Could not make it work. And I have read all of your suggestions, also on other threads, forums and sites. Where I found that, even the MRAA site suggest this. But it does not work (at least not for me). 

Tried the Yocto image (latest from Intel) and the Debian image (Ubilinux) . All with the same result. Even baked my own Yocto. Did not help either.

Maybe something is wrong with the Edision. But why is the MRAA sample working then?

rwaldron, 2015-01-12:
Can you paste your own program code? I am going to try to reproduce

Injac, 2015-01-12:
Sure. Here is the code I tested:

File 1:
```javascript
var five = require("johnny-five");
var Edison = require("edison-io");
var board = new five.Board({
  io: new Edison()
});

board.on("ready", function() {
  var led = new five.Led(13);
  led.blink();
});

```

File 2:

```javascript
// Use require("galileo-io") when running from an npm installation
var Galileo = require("galileo-io");
var board = new Galileo();

board.on("ready", function() {
  console.log("READY");
  var byte = 0;

  setInterval(function() {
    board.digitalWrite(13, (byte ^= 1));
  }, 500);
});
```

File 3:

```javascript
var five = require('johnny-five'),
    board;

board = new five.Board();

board.on('ready',function(){

  var led = new five.Led(13);
  led.blink(500);
});
```

This one works:

```javascript
var m = require('mraa'); //require mraa
console.log('MRAA Version: ' + m.getVersion()); //write the mraa version to the console

var myLed = new m.Gpio(13); //LED hooked up to digital pin 13 (or built in pin on Galileo Gen1 & Gen2)
myLed.dir(m.DIR_OUT); //set the gpio direction to output
var ledState = true; //Boolean to hold the state of Led

periodicActivity(); //call the periodicActivity function

function periodicActivity()
{
  myLed.write(ledState?1:0); //if ledState is true then write a '1' (high) otherwise write a '0' (low)
  ledState = !ledState; //invert the ledState
  setTimeout(periodicActivity,1000); //call the indicated function after 1 second (1000 milliseconds)
}
```

Hope this helps.

rwaldron, 2015-01-12:
I just ran all three Johnny-Five/Galileo-IO/Edison-IO examples with no issues (all of them make 13 blink); also ran the mraa-only example to confirm. 

- Edison connected to Arduino breakout shield
- edison-io@0.6.4
- galileo-io@0.6.4
- johnny-five@0.8.37
- mraa@0.5.4-71-g17ee0c9

What version of Galileo-IO and Johnny-Five do you have installed?

Injac, 2015-01-12:
Here are my versions:

* galileo-io@0.6.4
* edison-io@0.6.4
* johnny-five@0.8.37

But no mraa installed. npm list does not show it. Here is my outpu (sorry about that long thing here)t:

├── edison-io@0.6.4
├─┬ galileo-io@0.6.4
│ ├── es6-promise@2.0.1
│ ├── es6-shim@0.22.2
│ ├── graceful-fs@2.0.3
│ └─┬ remapped@0.2.1
│   ├── getobject@0.1.0
│   └── traverse@0.6.6
└─┬ johnny-five@0.8.37
  ├─┬ array-includes@1.0.4
  │ ├─┬ define-properties@1.0.1
  │ │ ├── foreach@2.0.5
  │ │ └── object-keys@1.0.3
  │ └── es-abstract@1.0.0
  ├── color-convert@0.5.2
  ├── colors@1.0.3
  ├── descriptor@0.1.0
  ├── ease-component@1.0.0
  ├── es6-shim@0.22.2
  ├─┬ firmata@0.4.0
  │ ├── browser-serialport@1.0.6
  │ └── object-assign@1.0.0
  ├── lodash@2.4.1
  ├── nanotimer@0.3.1
  ├─┬ serialport@1.4.10
  │ ├── async@0.9.0
  │ ├── bindings@1.2.1
  │ ├── nan@1.3.0
  │ ├─┬ node-pre-gyp@0.5.19
  │ │ ├─┬ mkdirp@0.5.0
  │ │ │ └── minimist@0.0.8
  │ │ ├─┬ nopt@2.2.1
  │ │ │ └── abbrev@1.0.5
  │ │ ├─┬ npmlog@0.0.6
  │ │ │ └── ansi@0.2.1
  │ │ ├─┬ rc@0.4.0
  │ │ │ ├── deep-extend@0.2.10
  │ │ │ ├── ini@1.1.0
  │ │ │ ├── minimist@0.0.10
  │ │ │ └── strip-json-comments@0.1.3
  │ │ ├─┬ request@2.36.0
  │ │ │ ├── aws-sign2@0.5.0
  │ │ │ ├── forever-agent@0.5.2
  │ │ │ ├─┬ form-data@0.1.2
  │ │ │ │ ├── async@0.2.10
  │ │ │ │ └─┬ combined-stream@0.0.4
  │ │ │ │   └── delayed-stream@0.0.5
  │ │ │ ├─┬ hawk@1.0.0
  │ │ │ │ ├── boom@0.4.2
  │ │ │ │ ├── cryptiles@0.2.2
  │ │ │ │ ├── hoek@0.9.1
  │ │ │ │ └── sntp@0.2.4
  │ │ │ ├─┬ http-signature@0.10.0
  │ │ │ │ ├── asn1@0.1.11
  │ │ │ │ ├── assert-plus@0.1.2
  │ │ │ │ └── ctype@0.5.2
  │ │ │ ├── json-stringify-safe@5.0.0
  │ │ │ ├── mime@1.2.11
  │ │ │ ├── node-uuid@1.4.1
  │ │ │ ├── oauth-sign@0.3.0
  │ │ │ ├── qs@0.6.6
  │ │ │ ├─┬ tough-cookie@0.12.1
  │ │ │ │ └── punycode@1.2.4
  │ │ │ └── tunnel-agent@0.4.0
  │ │ ├── rimraf@2.2.8
  │ │ ├── semver@2.3.0
  │ │ ├─┬ tar@0.1.19
  │ │ │ ├── block-stream@0.0.7
  │ │ │ ├─┬ fstream@0.1.25
  │ │ │ │ ├── graceful-fs@2.0.3
  │ │ │ │ └── mkdirp@0.3.5
  │ │ │ └── inherits@2.0.1
  │ │ └─┬ tar-pack@2.0.0
  │ │   ├── debug@0.7.4
  │ │   ├─┬ fstream@0.1.25
  │ │   │ ├── graceful-fs@2.0.3
  │ │   │ ├── inherits@2.0.1
  │ │   │ └── mkdirp@0.3.5
  │ │   ├─┬ fstream-ignore@0.0.7
  │ │   │ ├── inherits@2.0.1
  │ │   │ └─┬ minimatch@0.2.14
  │ │   │   ├── lru-cache@2.5.0
  │ │   │   └── sigmund@1.0.0
  │ │   ├── graceful-fs@1.2.3
  │ │   ├── once@1.1.1
  │ │   ├─┬ readable-stream@1.0.27-1
  │ │   │ ├── core-util-is@1.0.1
  │ │   │ ├── inherits@2.0.1
  │ │   │ ├── isarray@0.0.1
  │ │   │ └── string_decoder@0.10.25-1
  │ │   └── uid-number@0.0.3
  │ ├─┬ optimist@0.6.1
  │ │ ├── minimist@0.0.10
  │ │ └── wordwrap@0.0.2
  │ └── sf@0.1.7
  └── temporal@0.4.0

Do I need to install the mraa package separately?

rexstjohn, 2015-01-22:
I am getting the same problem with the LED refusing to blink. Tried the opkg update of mraa as well (version v0.5.4-100-g6c73a8a):

RROR:     state.io = new io.Gpio(this.addr);                                                                                                                                     
    ERROR:            ^                                                                                                                                                           
ERROR: Error: Illegal arguments for construction of _exports_Gpio                                                                                                                 
    at new Pin (/node_app_slot/node_modules/edison-io/node_modules/galileo-io/lib/mraa.js:66:16)                                                                                  
    at Galileo.<anonymous> (/node_app_slot/node_modules/edison-io/node_modules/galileo-io/lib/galileo.js:172:16)                                                                  
    at Array.map (native)                                                                                                                                                         
    at new Galileo (/node_app_slot/node_modules/edison-io/node_modules/galileo-io/lib/galileo.js:168:24)                                                                          
    at Object.<anonymous> (/node_app_slot/main.js:4:7)                                                                                                                            
    at Module._compile (module.js:456:26)                                                                                                                                         
    at Object.Module._extensions..js (module.js:474:10)                                                                                                                           
    at Module.load (module.js:356:32)                                                                                                                                             
    at Function.Module._load (module.js:312:12)                                                                                                                                   
    at Function.Module.runMain (module.js:497:10) 

Using:

var five = require("johnny-five");
var Edison = require("edison-io");
var board = new five.Board({
  io: new Edison()
});

board.on("ready", function() {
  var led = new five.Led(13);
  led.blink();
});

Package file I tried to use last:

{
  "name": "Johnny-Five",
  "description": "",
  "version": "0.0.0",
  "main": "main.js",
  "engines": {
    "node": ">=0.10.0"
  },
  "dependencies": {
      "johnny-five":"0.8.37",
      "edison-io":"0.6.4"
  }
}

rwaldron, 2015-01-23:
I will take a look first thing in the morning. Is this on an Edison or Galileo Gen 2?

rwaldron, 2015-01-23:
What happens when you run this without explicitly including the IO Plugin: 

```js
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  var led = new five.Led(13);
  led.blink();
});
```

rwaldron, 2015-01-23:
I need more info: 

- Galileo Gen 1 or Gen 2?
- Edison Arduino Board or Edison Mini Board?

rexstjohn, 2015-01-23:
@rwaldron Its an Edison Arduino board. I suspect it is MRAA-related and not johnny-five related. I will try this out in a bit.

Injac, 2015-01-23:
@rwaldron I have the Edison with the Mini Board.

Injac, 2015-01-24:
@rexstjohn Could you solve the problem? And how would one check on that? Specifically debug the node.js bindings to MRAA? How do you do that?

rwaldron, 2015-01-24:
@Injac I just updated to the latest version of mraa bindings and everything works correctly. Looking at the version you showed above, you've been using an out of date release: "v0.5.4-66-g62c3264". In my one of my earlier responses, I said I was using "0.5.4-71-g17ee0c9", which was already ahead of yours; now I'm on "v0.5.4-100-g6c73a8a". 

Please run the following: 

```
echo "src mraa-upm http://iotdk.intel.com/repos/1.1/intelgalactic" > /etc/opkg/mraa-upm.conf;
opkg update;
opkg install libmraa0;
node -e 'console.log(require("mraa").getVersion());'
```

This should end with "v0.5.4-100-g6c73a8a" (or newer)

rwaldron, 2015-01-24:
@Injac sorry, I missed where you said you were using the mini board. Testing that now.

rwaldron, 2015-01-24:
I've pinged the maintainers of mraa and will keep working once I hear from them.

Injac, 2015-01-25:
@rwaldron Thank you :)

rwaldron, 2015-01-25:
I'm about to spend some time working on this, will keep you posted

closed 2015-01-25 by rwaldron

rwaldron, 2015-01-26:
@Injac can you try version 0.7.0

Injac, 2015-01-26:
@rwaldron I will try tonight and let you know about the result.

Injac, 2015-01-26:
@rwaldron Works :D All three samples I posted. What was the issue?

rwaldron, 2015-01-26:
@Injac thanks for confirming :)

The issue was caused by the Edison Mini Board having a completely different set of pins from the Galileo (1, 2) and Edison Arduino Board. This table: https://github.com/intel-iot-devkit/mraa/blob/master/docs/edison.md#intelr-breakout-board shows which pins aren't actually connected (anything that says "nothing from mraa") which was causing the pre initialization to fail when it attempted to init pin 1

Injac, 2015-01-26:
I see. Thanks for the feedback and for fixing this issue so fast :) Made my day!

rwaldron, 2015-01-26:
My pleasure and thank _you_ for your patience :)

Injac, 2015-01-26:
You are very welcome :)