rwaldron/galileo-io #27: Devices undefined, REPL initiated error on Intel Edison

rexstjohn, 2015-04-04:
[image]

Using Johnny-Five with Intel Edison week 14 build (upgraded to latest LibMRAA). Getting this REPL issue. Any idea? The program stops and gives us this prompt and then Edison reboots if you hit "Enter."

rwaldron, 2015-04-04:
I just realized that Edison-IO was a version behind Galileo-IO. The version difference was the difference between a program fatal bug or not. First I updated so that the version was current. I wasn't able to reproduce the error, but I did notice that the version of mraa that Galileo-IO installs and depends on needed to be updated. I updated to `mraa@0.6.1-36-gbe4312e` and ran some tests, all successful: no crash and consistently correct behaviour. I've just updated the version and published to npm. 

----

If that doesn't solve the issue, can you provide the output of the following commands: 

```
uname -a;
cat /etc/version;
node -e "console.log(require('mraa').getVersion())";
```

Here's my current environment: 

```
root@rover:~# uname -a
Linux rover 3.10.17-poky-edison+ #1 SMP PREEMPT Fri Jan 30 14:16:35 CET 2015 i686 GNU/Linux
root@rover:~# cat /etc/version
weekly-120
root@rover:~# node -e "console.log(require('mraa').getVersion())"
v0.6.1-36-gbe4312e
```

- johnny-five@0.8.50
- galileo-io@0.8.3 (or edison-io@0.8.3)

rexstjohn, 2015-04-09:
What seems to happen is people get stuck at this point during install, then people hit CNTRL + C and try to use it and get that issue. Lets see if your above solution changes anything.

[image]

rexstjohn, 2015-04-09:
Still getting stuck on the node scripts/postinstall during "npm install." If I quit and run "node blink.js" I get one blink then the Repl. 

Here is my info: 

``` bash
root@rexison:~/.node_app_slot# uname -a
Linux rexison 3.10.17-poky-edison+ #1 SMP PREEMPT Fri Jan 30 14:16:35 CET 2015 i686 GNU/Linux
root@rexison:~/.node_app_slot# cat /etc/version
weekly-120
root@rexison:~/.node_app_slot# node -e "console.log(require('mraa').getVersion()
)"
v0.6.1
```
"johnny-five": "0.8.50",
"edison-io": "0.8.3"

rexstjohn, 2015-04-09:
Here is my package file:

``` bash
{
  "name": "edison-johnny-five",
  "description": "A simple Edison blink.",
  "version": "0.0.1",
  "main": "blink.js",
  "engines": {
    "node": ">=0.10.0"
  },
  "dependencies": {
      "johnny-five": "0.8.50",
      "edison-io": "0.8.3"
  }
}
```

rwaldron, 2015-04-09:
It's _not_ getting stuck—it's working. Let it finish!!

rwaldron, 2015-04-09:
> people get stuck at this point during install, 

It's not stuck, Galileo-IO is installing a known stable version of mraa to use, because I've learned that it's **really bad** to trust mraa releases. 

> then people hit CNTRL + C 

Don't do that. 

I will add some kind of message that says "don't ^C this, let it finish"

rexstjohn, 2015-04-09:
I'll just leave it someplace to run for awhile. People were having it sit there for a time and getting impatient. Crappy hackathon internet.

rexstjohn, 2015-04-09:
Left it for 20 minutes while I got tacos. Didn't complete by the time I got back. I hit "enter" and it exited. 

``` bash
node -e "console.log(require('mraa').getVersion())"
```

Returns "no libmraa installed."

rexstjohn, 2015-04-09:
Going to reflash this thing and try again.

rexstjohn, 2015-04-09:
Ok it works now. I did something stupid which was reflashing the Edison without setting the partition to FAT32 on the Mac. Once I did that, I did the following, now pin 13 blinks. Thanks for your help Rick you are awesome. Can't wait to teach all these students on this this weekend at HackUMASS. Deck has moved over here: http://slides.com/rexstjohn/deck-6-8.

Here is everything to make it work (assuming the partition is performed correctly):

``` bash
$ echo "src mraa-upm http://iotdk.intel.com/repos/1.1/intelgalactic" > /etc/opkg/mraa-upm.conf
$ opkg update
$ opkg install libmraa0
$ opkg upgrade
$ npm install
$ node blink.js
```

``` bash
root@rexison:~/.node_app_slot# uname -a
Linux rexison 3.10.17-poky-edison+ #1 SMP PREEMPT Fri Jan 30 14:16:35 CET 2015 i686 GNU/Linux
root@rexison:~/.node_app_slot# cat /etc/version
weekly-120
root@rexison:~/.node_app_slot# node -e "console.log(require('mraa').getVersion()
)";
v0.6.1-36-gbe4312e
```

package.json

``` bash
{
  "name": "edison-johnny-five",
  "description": "A simple Edison blink.",
  "version": "0.0.1",
  "main": "blink.js",
  "engines": {
    "node": ">=0.10.0"
  },
  "dependencies": {
      "johnny-five": "0.8.5",
      "edison-io": "0.8.3"
  }
}
```

blink.js

``` bash
// Blink an LED
var five = require("johnny-five");
var Edison = require("edison-io");
var board = new five.Board({
  io: new Edison()
});

board.on("ready", function() {
  var led = new five.Led(13);
  led.blink(1000);
});
```

rwaldron, 2015-04-09:
This is great news :)

I'm going to land some kind of "progress" indicator as a patch to close this. 

Rick

ashishdatta, 2015-04-10:
@rwaldron could we possibly have a check to see if the libmraa installed is a known "good" release? That way we avoid having to fetch unnecessarily. Something maybe like this? https://gist.github.com/ashishdatta/e1c5086b716de1ea5f5b

rwaldron, 2015-04-10:
Yeah, that's a great idea :)

rwaldron, 2015-04-10:
Landed https://github.com/rwaldron/galileo-io/commit/5ea58b636d98ef0bb9294c241ee588d2a065c551

rwaldron, 2015-04-10:
@rexstjohn you'll want to update your materials to reference v0.8.4

@ashishdatta thanks for the help/idea!

closed 2015-04-10 by rwaldron