rwaldron/galileo-io #28: Issue creating a new sensor on pin A0

scazan, 2015-04-26:
I can't seem to initialize pin A0 when using this on the Galileo 2. I keep getting an error about pin A14. Is there a different way I should be initializing these? Thanks!

Here is what I run:
```
var five = require('johnny-five');
var board = new five.Board({repl: false});

board.on("ready", function() { console.log('board ready'); });

var sensor = five.Sensor({
pin: "A0",
freq: 100
});
```

And here is the error I get back at that point:
```
Error: Edison Mini Board has no connection at pin A14
    at ToPinIndex (/home/root/server/node_modules/johnny-five/node_modules/galileo-io/lib/galileo.js:219:9)
    at Galileo.pinMode (/home/root/server/node_modules/johnny-five/node_modules/galileo-io/lib/galileo.js:333:14)
    at new Sensor (/home/root/server/node_modules/johnny-five/lib/sensor.js:62:11)
    at Object.Sensor (/home/root/server/node_modules/johnny-five/lib/sensor.js:37:12)
    at repl:1:19
    at REPLServer.self.eval (repl.js:110:21)
    at repl.js:249:20
    at REPLServer.self.eval (repl.js:122:7)
    at Interface.<anonymous> (repl.js:239:12)
    at Interface.EventEmitter.emit (events.js:95:17)
```

scazan, 2015-05-05:
This looks like an issue with using five.Sensor specifically. It would seem the pin offset adds upon itself. Pin A0 becomes 14 which is then returned to Sensor at some point, then is reprocessed as A14 which results in an out of bounds pin index of 28. 

I changed my code to work via analogRead directly for the time being.

rwaldron, 2015-05-05:
Thanks for the report!

First thing, this example is invalid: 

```js
var five = require('johnny-five');
var board = new five.Board({repl: false});

board.on("ready", function() { console.log('board ready'); });

var sensor = five.Sensor({
pin: "A0",
freq: 100
});
```

Component initialization _must_ happen within the board's "ready" handler to ensure that the proper initializations have been performed. 

Try this: 

```js
var five = require('johnny-five');
var board = new five.Board({repl: false});

board.on("ready", function() { 
  var sensor = five.Sensor({
    pin: "A0",
    freq: 100
  });
});
```

If this doesn't work, then I'll need to know what OS image is running on your Galileo Gen2. It may also be that the native bindings are detecting the board as though it were an Intel Edison connected to the Mini Board.

scazan, 2015-05-05:
Ah yes. I was putting directly into the REPL so that I would not run the sensor code until I received that "board ready" message (making sure that all inits are done beforehand).

I'm running the same sensor code as part of the callback in my project and getting the same results.

I am running an Intel IoT Devkit image. I might try replacing the OS image with a newer one. When I installed this one there were a whole lot of different OS images out there and no clear indication from Intel as to which one was best so I chose the most official looking download.

rwaldron, 2015-05-05:
I have the latest IoT DevKit image on my Galileo Gen2, can you update so we have the same environment?

> Ah yes. I was putting directly into the REPL so that I would not run the sensor code until I received that "board ready" message (making sure that all inits are done beforehand).

Do you mean that you're starting a node REPL like this? 

```
$ node
>  //  then do stuff here...
```

rwaldron, 2015-05-05:
Ohhhh, also I noticed that you're not including the required IO Plugin—yikes! Try this: 

```
npm install galileo-io
```

Then... 

```js
var five = require("johnny-five");
var Galileo = require("galileo-io");
var board = new five.Board({
  io: new Galileo()
});

board.on("ready", function() {
  var sensor = new five.Sensor({
    pin: "A0",
    freq: 100
  });

  sensor.on("data", function() {
    console.log(this.value);
  });
});
```

scazan, 2015-05-06:
Hmm.... Still getting the same results. Johnny-Five was autodetecting and loading the galileo-io plugin before so I wasn't including it (I verified that it had the latest version as well).

I downloaded a fresh version of the SD Card image from Intel (https://software.intel.com/en-us/iot/hardware/galileo/downloads), installed johnny-five and galileo-io, and then copied and pasted the code above into a test.js file which I then ran and got these results:

```
node test.js 
1424914115157 Device(s) Intel Galileo Gen 2 
1424914115224 Connected Intel Galileo Gen 2 
1424914115232 Repl Initialized 
>> 
Error: Edison Mini Board has no connection at pin A14
    at ToPinIndex (/home/root/server/node_modules/galileo-io/lib/galileo.js:240:9)
    at Galileo.pinMode (/home/root/server/node_modules/galileo-io/lib/galileo.js:359:14)
    at new Sensor (/home/root/server/node_modules/johnny-five/lib/sensor.js:62:11)
    at Board.<anonymous> (/home/root/server/test.js:8:16)
    at Board.emit (events.js:95:17)
    at Board.broadcast (/home/root/server/node_modules/johnny-five/lib/board.js:395:8)
    at Board.<anonymous> (/home/root/server/node_modules/johnny-five/lib/board.js:309:23)
    at Galileo.g (events.js:180:16)
    at Galileo.emit (events.js:92:17)
    at Galileo.<anonymous> (/home/root/server/node_modules/galileo-io/lib/galileo.js:302:10)
```

Are you able to access A0? Odd that it is giving these results.

rwaldron, 2015-05-06:
Ok, I figured this out and I'm super sorry that I didn't address it sooner :|

Here's what happened: When I added support for the Sparkfun blocks, I wanted the user to be able to use the pin names as they are printed on the block, which meant using the IO-Plugin `normalize(pin)` to expose `ToPinIndex(pin)`. I _thought_ I had proper tests for this, but I clearly overlooked this case. I will have a patch shortly.

Thanks again for your patience :)

closed 2015-05-06 by rwaldron

rwaldron, 2015-05-06:
Please update to galileo-io@0.8.12

- I was able to produce the issue consistently
- Patched, with lots of new tests and updated tests
- Confirmed working correctly on 
    - Intel Edison Arduino Board, Intel Galileo Gen 2
    - Intel Edison Mini Board
    - Intel Edison Sparkfun GPIO block
    - Intel Edison Sparkfun Arduino block

rwaldron, 2015-05-06:
Thanks again!!

scazan, 2015-05-06:
Thank YOU! Really great to have this.