rwaldron/galileo-io #29: Steppers not supported?

rubenmp8, 2015-04-27:
I am trying to create a stepper control on an Edison board by doing the following taken from the johnny-five docs

  stepper = new five.Stepper({
    type: five.Stepper.TYPE.DRIVER,
    stepsPerRev: 200,
    pins: {
      step: 11,
      dir: 13
    }
  });

However I am getting a Stepper is not supported error

Error: Stepper is not supported
    at new Stepper (/home/rmperez/test/node_modules/johnny-five/lib/stepper.js:128:11)
    at initializeStepper (/home/rmperez/test/app/controllers/mcu.js:28:13)
    at Board.exports.gotoPosition (/home/rmperez/test/app/controllers/mcu.js:41:3)

I made sure to upload a sketch using AdvancedFirmata before this. Are steppers not supported in galileo-io or am I doing something incorrectly? Thanks

rwaldron, 2015-04-27:
Thanks for the report, unfortunately...

> I am trying to create a stepper control on an Edison board by doing the following taken from the johnny-five docs.
> ...
> However I am getting a Stepper is not supported error

This is correct. The docs also include this notice: 

> In order to use the Stepper class, your board must be flashed with AdvancedFirmata, which is available here: https://github.com/soundanalogous/AdvancedFirmata

(And the platform page: http://johnny-five.io/platform-support/#intel-edison-arduino)

I apologize for this being unclear, and I have updated that to be more specific: 

> Currently, `Stepper` is only supported when using Johnny-Five with an Arduino (tethered via serial usb). In order to use the `Stepper` class, your board must be flashed with `AdvancedFirmata`, which is available here: https://github.com/soundanalogous/AdvancedFirmata

Please note that this is only temporary. Updates to follow

rwaldron, 2015-04-27:
Also, you'll want to remove that sketch from the system: 

```
rm /sketch/sketch.elf 
```

rwaldron, 2015-04-27:
As it turns out, if you're using the SparkFun Arduino Block, you can flash Advanced Firmata to the Arduino and run JS scripts from the Edison. https://github.com/rwaldron/galileo-io#sparkfun-edison-arduino-block

rubenmp8, 2015-04-27:
Cool, I have the arduino breakout board, but I'll look into using the Sparkfun block as a temporary workaround.

Out of curiosity, what are the issues blocking Stepper support on edison+arduino breakout board?

rwaldron, 2015-04-27:
> Out of curiosity, what are the issues blocking Stepper support on edison+arduino breakout board?

None, I just haven't finished writing support yet :)

I worked on it all day today, should be ready for testing tomorrow-ish.

rubenmp8, 2015-04-27:
Awesome, thanks a bunch for working on this immediately!

rwaldron, 2015-04-27:
Sometimes I just need a little outside motivation ;) 

Appreciate you bringing this bug :)

rwaldron, 2015-05-01:
Update: 

- Stepper support is nearly there
- Speed is a problem and has been the cause of delays

rwaldron, 2015-05-08:
No matter what I try, I cannot get the speed to increase beyond a certain point.

rubenmp8, 2015-05-08:
Ouch, how slow are we talking?