rwaldron/galileo-io #30: Issue w/ setSamplingInterval in J5, galileo-io, node-red and node-red-contrib-gpio setup

Jbossalini, 2015-05-01:
re: https://github.com/rwaldron/galileo-io/issues/19 Implement setSamplingInterval

I'm reading a TMP36 temp sensor connected to A0 of an Edison plugged into an Arduino breakout board, using Node-red and the `node-red-contrib-gpio` input node.  I can read the sensor, but the `setSamplingInterval` value is not being honored, and the debug output shows that the sensor is being read 100s of times a second.  When I start up Node-red, I see this error in the log:

```
[TypeError: Object #<Galileo> has no method 'setSamplingInterval']
```

Is `setSamplingInterval` implemented in galilieo-io yet, or is the error coming from somewhere else?  I'm using v 0.8.8 of `galileo-io`.

Thanks!  -Joe

rwaldron, 2015-05-01:
Thanks for the report. This should be all set now

closed 2015-05-01 by rwaldron

rwaldron, 2015-05-01:
Also, just wanted to make sure that you're using the `Temperature` class for the TMP36: http://johnny-five.io/api/temperature

Jbossalini, 2015-05-05:
@rwaldron Thanks!  Error in Node-red is gone, and I can run the J5 sample script (with interval). Thanks for the quick turnaround!

I've now got another issue in Node-red where I'm not getting any readings at all, so I've temporarily reverted to a previous `galileo-io` version and hard coded the interval.  I've got a deadline today for a project I'm using it with, so I'll dig back into it some more at the end of the week.  Might be caused by `node-red-contrib-gpio` but I won't know until I can debug further.