rwaldron/galileo-io #31: trouble to install galileo-IO

alperpelik, 2015-05-03:
is there any issue about installation. I am trying to install io. but it is waiting very very long time ( like 2-3hours) and it is not completely installing.

My board is Galileo-Gen2.

Alper

scazan, 2015-05-05:
Is this the point at which it says it is installing a safe version of mraa? I had that problem as well. It would simply hang at that point (with a warning not to restart). 

I worked around it by Ctrl-C'ing at that point, going into the johnny-five directory and running this manually:

```
npm install mraa@0.6.1-36-gbe4312e
```

If that still fails you can try the unsafe version by running:
```
npm install mraa
```

But I assume that is probably a bad idea.

rwaldron, 2015-05-05:
What OS image do you have on your Galileo Gen2?

scazan, 2015-05-06:
I was able to repeat this issue on the latest IoT image downloaded from Intel. I can also verify that it did actually install libmraa. It just didn't seem to indicate that it had finished (or output any of the opkg stdout).

Running Galileo-IO after terminating the postinstall.js seemed to work fine.

rwaldron, 2015-05-06:
I did fresh installs of Galileo-IO on 3 different boards today: 2 different Edisons and a Galileo Gen 2; no issues. 

> Running Galileo-IO after terminating the postinstall.js seemed to work fine.

I'll look into further improving the output/results of this installation.

alperpelik, 2015-05-07:
Yes. I did it. I thing something was missing on my sd-image. I reload image to Sdcard and try it. I worked.I did  not understand why but now it is working.

rwaldron, 2015-05-07:
@alperpelik thanks for the update

@scazan can you try installing 0.8.13, thanks!