rwaldron/galileo-io #33: Device or Firmware Error

rubenmp8, 2015-06-04:
I am trying to use johnny-five with the SparkFun Edison Arduino Block, but am getting a "Device or Firmware Error A timeout occurred while connecting to the Board" error upon trying the blink example from the galileo-io docs for the arduino block. I used the FTDI breakout to upload standardfirmata, and ran the edison client example from https://learn.sparkfun.com/tutorials/sparkfun-blocks-for-intel-edison---arduino-block. The example does cause pin 13 to blink, so it would seem firmata is indeed correctly installed onto the arduino itself. Any suggestions for what might be wrong?

rwaldron, 2015-06-04:
Thanks for the report! Let's see if we can get you up and running. A couple things: 

- when using the Sparkfun Edison Arduino block, Johnny-Five doesn't need Galileo-IO (in fact, it just won't work). 
- Did you explicitly specify the serial port path when creating the Board instance? node-serialport's `list` won't auto detect the port

rubenmp8, 2015-06-04:
I am only requiring johnny-five, but I figured people here might have direct experience with the board. I am using /dev/ttyMFD1 for the port. The full code I am using follows

```
var five = require("johnny-five");
var board = new five.Board({
port:"/dev/ttyMFD1"
});

board.on("ready", function() {
        var led = new five.Led(13);
        led.blink(500);
});
```