rwaldron/galileo-io #35: What do you think about separate IO plugins for the various Edison boards?

codefoster, 2015-06-14:
@rwaldron, what do you and the rest of the j5 community think about separate modules for the various Edison boards - Arduino dev board, miniboard, Sparkfun board, and the new Seeed Xadow board?
I recently spent some time trying to get the ADXL345 working with the Xadow board and discovered that it uses the 0 I2c bus (see [galileo.js](https://github.com/rwaldron/galileo-io/blob/master/lib/galileo.js) at line 503). I don't see a platform query that can be done to determine that the Xadow is what's being used, so I wonder if it's best to branch by file.
The advantage is being able to lose the branching code (there's not a ton of it). The disadvantage is obviously the duplicate code and the fact that a developer has to differentiate between the plugins when he's learning the platform.
Opinions?

ashishdatta, 2015-06-14:
My $0.02:

Currently (just reading the readme) @rwaldron patch to fix this is to declare the I2c bus:

```
var five = require("johnny-five");
var Edison = require("galileo-io");
var board = new five.Board({
  io: new Edison({ bus: 0 })
});
``` 

As far as I can tell the only difference between these expansion boards on a software level is that they use different i2c buses and have different number of pins (please correct me if I'm wrong). While it is nice to be able to declare what i2c bus I want to use as this will help if I am creating my own expansion board, the average everyday user might not understand even what an i2c bus is! 

My suggest would be to just declare the expansion board we are going to use instead like:

```
var five = require("johnny-five");
var Edison = require("galileo-io");
var board = new five.Board({
  io: new Edison(seeed_xadow_board)
});
``` 
And we do the changes as needed in the background. Either way this is an important issue as I see more and more of these expansion boards coming out.

rwaldron, 2015-06-15:
Thanks for the proposal. The inevitable maintenance burden of dividing one IO Plugin into four (or more) is unjustifiable. 

However, I am convinced that @ashishdatta's proposal, which serves the same purpose is a valuable addition. I'm going to work on fleshing out a "board schema" today.

codefoster, 2015-06-15:
Love it.

dtex, 2015-06-15:
That is essentially what we did for Johnny-Five motor shields. Predefined opts for different shield types. Very nice for the user and super easy to maintain.

rwaldron, 2015-06-16:
I committed a draft yesterday, but decided I didn't like that approach, so I scaled it back to the much simpler version that I've just pushed out in 0.8.16. Testing is greatly appreciated!

```js
new Galileo(Galileo.Boards.Xadow);
```

codefoster, 2015-06-16:
Thanks a billion. Will test and report.