rwaldron/galileo-io #36: install failure with status 255

jzheng-stoneriver, 2015-06-14:
I tried to do npm -g install galileo-io in Intel Edison with expansion board. It failed with the following message:

npm ERR! galileo-io@0.8.15 postinstall: `node scripts/postinstall`
npm ERR! Exit status 255
npm ERR!
npm ERR! Failed at the galileo-io@0.8.15 postinstall script.
npm ERR! This is most likely a problem with the galileo-io package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node scripts/postinstall
npm ERR! You can get their info via:
npm ERR!     npm owner ls galileo-io
npm ERR! There is likely additional logging output above.
npm ERR! System Linux 3.10.17-poky-edison+
npm ERR! command "node" "/usr/bin/npm" "-g" "install" "galileo-io@latest"
npm ERR! cwd /home/root
npm ERR! node -v v0.10.38
npm ERR! npm -v 1.4.28
npm ERR! code ELIFECYCL

ashishdatta, 2015-06-14:
Hi! 

I am able to reproduce this error when I try to install on my board as well, but if I ran it without the -g meaning ```npm install galileo-io``` it worked fine, I was even able to install in /usr/lib/node_modules. 

My wild guess is that there is something weird with installing globally and using the ```exec``` command from ```postinstall.js```

So my suggestion is to either install it locally to your project or do these commands to get it globally:

```
cd /usr/lib/node_modules
npm install galileo-io
```

Test to make sure it worked: 

```
cd ~/
root@saber:~# node
> var x = require('galileo-io')
undefined
```

ashishdatta, 2015-06-18:
@jzheng-stoneriver did this resolve the issue?

jzheng-stoneriver, 2015-06-18:
I could confirm the local installation worked. I am fine to use this as a workaround.

closed 2015-06-18 by jzheng-stoneriver

ashishdatta, 2015-06-20:
@rwaldron any insight as to why when we invoke ```-g``` in ```npm install``` we get this error?