rwaldron/galileo-io #38: [question] Any chance of supporting debian?

Awk34, 2015-06-27:
I'm trying to run my Edison with Debian. Attempting to install fails when it tries to make `opkg` commands.

```
edison@ubilinux:~/$ sudo npm i -g galileo-io
\
> galileo-io@0.8.18 postinstall /usr/lib/node_modules/galileo-io
> node scripts/postinstall

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
  Do not quit the program until npm completes the installation process.
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
{ [Error: Command failed: /bin/sh -c opkg info libmraa0
/bin/sh: 1: opkg: not found
]
  killed: false,
  code: 127,
  signal: null,
  cmd: '/bin/sh -c opkg info libmraa0' }
npm ERR! Linux 3.10.17-yocto-standard-r2
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "i" "-g" "galileo-io"
npm ERR! node v0.12.5
npm ERR! npm  v2.11.2
npm ERR! file sh
npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn

npm ERR! galileo-io@0.8.18 postinstall: `node scripts/postinstall`
npm ERR! spawn ENOENT
npm ERR!
npm ERR! Failed at the galileo-io@0.8.18 postinstall script 'node scripts/post  stall'.
npm ERR! This is most likely a problem with the galileo-io package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node scripts/postinstall
npm ERR! You can get their info via:
npm ERR!     npm owner ls galileo-io
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/edison/mraa-0.7.2/build/npm-debug.log
```

ashishdatta, 2015-06-27:
The reason this is failing is because the postinstall script is thinking
you are using a Yocto based system and thus tries to execute opkg commands
, whereas Ubilinx uses apt-get i believe.

I do not have a Ubilinx system but if you could run these commands I can
submit a fix.

node (starts the repl)
var os = require('os')
console.log(os.release)

Please post the results of the console.log, thanks!

Awk34, 2015-06-27:
For some strange reason `console.log(os.release())` spits out `3.10.17-yocto-standard-r2`, but I'm on Debian I swear!! Is this a problem with Ubilinux?

Yeah Debian uses `apt-get`

ashishdatta, 2015-06-27:
That is so strange, I will ask the ubilinux guys if we can fix that. What does os.platform() give you? Also if you wouldnt mind running 'cat /etc/version'

Awk34, 2015-06-27:
```js
> require('os').platform()
'linux'
```

Don't have an `/etc/version/`, but I have this:

```
edison@ubilinux:~/test$ cat /proc/version
Linux version 3.10.17-yocto-standard-r2 (root@p32mac) (gcc version 4.8.2 (GCC) ) #7 SMP PREEMPT Thu Feb 26 09:57:06 UTC 2015
```

ashishdatta, 2015-06-27:
Well that is disappointing, there doesnt seem to be mention of debian/ubilinix anywhere ... I believe os.release() uses the uname -a command so no point in trying that. If there was just some way of determining if you are on ubilinx this would be so much easier. 

This is probably the extend i can help with until I get a board with Ubilinx on it.

Awk34, 2015-06-27:
Yeah, so strange. Can I clone this repo and edit the `postinstall.js` to just pass on through and do an `npm link`? I already have `mraa` installed globally

Awk34, 2015-06-27:
Alright so I did that (cloned, removed postinstall from package.json, `sudo npm link`) and it seemed to install alright. Let me run a test...

ashishdatta, 2015-06-27:
I will send a note to them, wouldn't hurt if you sent one as well :)

Awk34, 2015-06-27:
Can you tell me what running `cat /etc/*-release` on a Yocto board outputs?

ashishdatta, 2015-06-27:
Can't seem to find my board. Will update when I can.

yocto1, 2015-07-13:
Having the same issues here, any progress since?

Awk34, 2015-07-13:
The PR I mentioned is still open. You can copy down that code and run `npm install` and `npm link` in it's root, and try to install the module again.

closed 2016-03-24 by Awk34