rwaldron/galileo-io #40: Can't seem to install galileo-io on latest edison image

gwfong, 2015-09-28:
I just reflashed a new intel edison image and configure wifi and whatnot. I can install other node packages, like johnny-five. But when it comes to galileo-io (or edison-io), I always get an error. Any ideas how I might be able to resolve this?

When I run:
npm install -g galileo-io

I get the following error:
npm ERR! galileo-io@0.8.18 postinstall: `node scripts/postinstall`
npm ERR! Exit status 255
npm ERR! 
npm ERR! Failed at the galileo-io@0.8.18 postinstall script.
npm ERR! This is most likely a problem with the galileo-io package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node scripts/postinstall
npm ERR! You can get their info via:
npm ERR!     npm owner ls galileo-io
npm ERR! There is likely additional logging output above.
npm ERR! System Linux 3.10.17-poky-edison+
npm ERR! command "node" "/usr/bin/npm" "install" "-g" "galileo-io"
npm ERR! cwd /home/root
npm ERR! node -v v0.10.38
npm ERR! npm -v 1.4.28
npm ERR! code ELIFECYCLE
npm ERR! not ok code 0

Thanks
Gary

gwfong, 2015-09-28:
Sorry. I found the workaround (https://github.com/rwaldron/galileo-io/issues/36). Thanks

closed 2015-09-28 by gwfong

rwaldron, 2015-09-28:
What version of node/npm did you hit this on? Also what is the version number for "latest edison image"?

gwfong, 2015-09-29:
node version: v0.10.38
npm version: 1.4.28
edison version: 159

The edison image was obtained from the all-in-one installer for Mac OS X. I got the installer from https://software.intel.com/en-us/iot/hardware/edison/downloads. After re-installing the new image, I used XDK to update the on-board libraries and the xdk-daemon before trying to install galileo-io (or edison-io).