rwaldron/galileo-io #43: Edison Mini Board has no connection at pin A

Legendaire, 2015-12-19:
var five = require("johnny-five");
var Edison = require("edison-io");

var board = new five.Board({
  io: new Edison()
});

board.on("ready", function() {

    console.log("board ready");
   
    var motorA = new five.Motor({controller: "GROVE_I2C_MOTOR_DRIVER", pin: "A", });
    var motorB = new five.Motor({controller: "GROVE_I2C_MOTOR_DRIVER", pin: "B", });
 
results in this:

1450486801662 Device(s) Intel Edison 
1450486801672 Connected Intel Edison                                                                                                                                                                                                                                 
1450486801690 Repl Initialized                                                                                                                                                                                                                                       
>> board ready                                                                                                                                                                                                                                                       
                                                                                                                                                                                                                                                                     
ERROR: Error: Edison Mini Board has no connection at pin A                                                                                                                                                                                                           
    at ToPinIndex (/home/root/.node_app_slot/node_modules/edison-io/node_modules/galileo-io/lib/galileo.js:332:13)                                                                                                                                                   
    at Galileo.normalize (/home/root/.node_app_slot/node_modules/edison-io/node_modules/galileo-io/lib/galileo.js:345:12)                                                                                                                                            
    at normalizer (/home/root/.node_app_slot/node_modules/johnny-five/lib/board.pins.js:109:35)                                                                                                                                                                      
    at Function.Pins.normalize (/home/root/.node_app_slot/node_modules/johnny-five/lib/board.pins.js:124:18)                                                                                                                                                         
    at Motor.Board.Component (/home/root/.node_app_slot/node_modules/johnny-five/lib/board.js:839:23)                                                                                                                                                                
    at new Motor (/home/root/.node_app_slot/node_modules/johnny-five/lib/motor.js:497:19)                                                                                                                                                                            
    at Board.<anonymous> (/home/root/.node_app_slot/main.js:25:18)                                                                                                                                                                                                   
    at Board.emit (events.js:92:17)                                                                                                                                                                                                                                  
    at process._tickCallback (node.js:448:13)                                                                                                                                                                                                                        
    at Function.Module.runMain (module.js:499:11)

rwaldron, 2015-12-19:
I believe this was resolved in https://github.com/rwaldron/galileo-io/commit/f737e8739cf4c4766bc46bdb90986aa716557381

Are you using this with the Arduino Breakout or Mini Breakout? If it's the Mini, how are you connecting the grove motor driver?

rwaldron, 2015-12-19:
Can you confirm both:

- Version of Johnny-Five
- Version of Edison/Galileo-io

rwaldron, 2015-12-19:
From the Gitter channel: 

> I am using the sparkfun console block as my base.

How is that connected to the Grove Motor Driver?

Legendaire, 2015-12-20:
I am using the Sparkfun Base block.  I misspoke about the console block earlier.  I was using whatever the latest edison-io was and it was giving the error.  I wiped and flashed one of my Edisons only to get the same error again.  I switched to galileo-io and the error went away.

rwaldron, 2015-12-21:
Well, Edison-IO is just an alias for Galileo-IO ;)

Legendaire, 2015-12-21:
I wonder why it didn't work.  Odd.

rwaldron, 2015-12-21:
Ok, but this error is no longer an issue?

Legendaire, 2015-12-21:
I believe it is working now with galileo-io.  I have not retested edison-io
and have no idea why it wouldn't have worked to begin with.  I'd say it's
fixed.  If not, we can always reopen the case.

rwaldron, 2015-12-21:
cool

closed 2015-12-21 by rwaldron