rwaldron/galileo-io #44: Servo on DFRobot Edison shield

Legendaire, 2015-12-21:
var five = require("johnny-five");
var Edison = require("galileo-io");

var board = new five.Board({
  io: new Edison()
});

board.on("ready", function() {
    console.log("board ready");
    var servPan = new five.Servo(5);

Results in this...

1450680905444 Device(s) Intel Edison                                                                                                                                                                                                                                 
1450680905453 Connected Intel Edison                                                                                                                                                                                                                                 
1450680905470 Repl Initialized                                                                                                                                                                                                                                       
>> board ready                                                                                                                                                                                                                                                       
                                                                                                                                                                                                                                                                     
ERROR: TypeError: Cannot read property '0' of undefined                                                                                                                                                                                                              
    at ToPinIndex (/home/root/.node_app_slot/node_modules/galileo-io/lib/galileo.js:304:19)                                                                                                                                                                          
    at Galileo.pinMode (/home/root/.node_app_slot/node_modules/galileo-io/lib/galileo.js:357:14)                                                                                                                                                                     
    at Servo.Controllers.Standard.initialize.value (/home/root/.node_app_slot/node_modules/johnny-five/lib/servo.js:59:19)                                                                                                                                           
    at new Servo (/home/root/.node_app_slot/node_modules/johnny-five/lib/servo.js:189:8)                                                                                                                                                                             
    at Board.<anonymous> (/home/root/.node_app_slot/main.js:46:19)                                                                                                                                                                                                   
    at Board.emit (events.js:92:17)                                                                                                                                                                                                                                  
    at process._tickCallback (node.js:448:13)                                                                                                                                                                                                                        
    at Function.Module.runMain (module.js:499:11)                                                                                                                                                                                                                    
    at startup (node.js:119:16)                                                                                                                                                                                                                                      
    at node.js:935:3   

I remembering someone saying the DFRobot board acts like the Intel Mini Breakout Board.  So my assumption is that one may not address the pins in the same way for the DFRobot board as they would for the Intel Adruino Breakout Board.  Would this be correct?  If so, is there a way to make it so one may address things the same way as on the Intel board?  If not, how does one address the pins on the DFRobot board?

rwaldron, 2015-12-21:
Are you referring to the DFRobot IO Expansion board? I have a WIP branch that does a bunch of pin mapping to make this is all nicer—I'll get that finished and landed

Legendaire, 2015-12-21:
Luckily I have both boards from them.  I would be able to test either.

rwaldron, 2015-12-21:
I've been working on this all day. One major issue I've come across is that PWM doesn't seem to actually function correctly on this board. I even tried controlling it with a sketch uploaded via Arduino IDE and it did not work.

rwaldron, 2015-12-21:
Also, this board crashes a lot :\

rwaldron, 2015-12-21:
And gets very hot

rwaldron, 2015-12-21:
Ok, strike all that, I tried with external power plugged in and it works "ok-ish". I think this is a poorly assembled board.

Legendaire, 2015-12-22:
I noticed a squirrelly response from the servos when I used GPIO pins such
as GP44 (I think).  I did not use external power.

rwaldron, 2015-12-22:
I always use external power for servos, otherwise the resulting behavior is unreliable