rwaldron/galileo-io #47: Johnny five + Intel Edison + BMP180

teuteuguy, 2016-03-08:
When trying to launch a simple barometer read it fails on:
RangeError: Trying to access beyond buffer length
    at checkOffset (buffer.js:582:11)
    at Buffer.readUInt8 (buffer.js:588:5)
    at Galileo.<anonymous> (/home/root/tests/barometer/node_modules/edison-io/node_modules/galileo-io/lib/galileo.js:696:24)
    at Timer.listOnTimeout [as ontimeout] (timers.js:112:15)

rwaldron, 2016-03-08:
Thanks for the report! This error will occur when the expected I2C peripheral (in this case, the BMP180) isn't found—which could be a wiring issue or just plain not attached. I agree that this is a crummy error. Can you tell me more about your circuit? In the meantime, I will add some better output for this

teuteuguy, 2016-03-12:
Hi rwalton,

unfortunately, you closed this a bit fast, cause the device IS connected via I2C to the board, and wiring is correct...

Intel Edison + Arduino Grove shield + Grove BMP180

rwaldron, 2016-03-12:
Yes, I used the exact same setup. Can you install the latest version and test again?

rwaldron, 2016-03-12:
> unfortunately, you closed this a bit fast

Sorry, github does this automatically when i include "Fixes gh-XXX" in the commit message. I will reopen until you confirm.

teuteuguy, 2016-03-14:
Nope, no success. 
Still getting errors:

1457926875390 Device(s) Intel Edison
1457926875405 Connected Intel Edison
1457926875429 Repl Initialized
>>
RangeError: Trying to access beyond buffer length
    at checkOffset (buffer.js:582:11)
    at Buffer.readUInt8 (buffer.js:588:5)
    at Galileo.<anonymous> (/home/root/tests/barometer/node_modules/edison-io/node_modules/galileo-io/lib/galileo.js:696:24)
    at Timer.listOnTimeout [as ontimeout] (timers.js:112:15)

teuteuguy, 2016-03-14:
Investigated a little further.

Note 1: this fix is for galileo-io, which is depended in, by edison-io (which is the module you need, I believe) for running Johnny-five on Edison.
So, I've gone into the node-modules/edison-io, updated the package.json to reflect the 0.9.3 version of galileo-io, npm installed.
Rerunning, seems to "fix" the issue. (However new issue now).
So regarding note1: I believe you probably have to update edison-io module to reflect the 0.9.3 change in galileo-io.

Note 2: now that note1 fixed it, I get a different issue.
`--------------------------------------
I2C: Could not read 3 Bytes from peripheral with address 0x77
I2C: Could not read 3 Bytes from peripheral with address 0x77
I2C: Could not read 2 Bytes from peripheral with address 0x77
I2C: Could not read 2 Bytes from peripheral with address 0x77
barometer
  pressure :  0.2360
--------------------------------------
barometer
  pressure :  0.2360
--------------------------------------
I2C: Could not read 3 Bytes from peripheral with address 0x77
I2C: Could not read 3 Bytes from peripheral with address 0x77
I2C: Could not read 2 Bytes from peripheral with address 0x77
I2C: Could not read 2 Bytes from peripheral with address 0x77`

in a never ending loop.

My code:
`var five = require("johnny-five");
var Edison = require("edison-io");
var board = new five.Board({
  io: new Edison()
});

board.on("ready", function() {

  var barometer = new five.Barometer({
    controller: 'BMP180'
  });

  barometer.on('data', function() {
    console.log("barometer");
    console.log("  pressure : ", this.pressure);
    console.log("--------------------------------------");
  });

});`

rwaldron, 2016-03-14:
Sorry, I did miss the dependency update in Edison-IO, that's fixed. 

> Note 2: now that note1 fixed it, I get a different issue (below is the output)

This error is _only_ possible if the device being read is not actually connected to the board. There is nothing this library can do to fix that, as it's simply forwarded up from native bindings. The board itself is saying "I can't find this thing you want me to read, sorry". 

Can you post a picture or two of your hardware as it's presently connected?

teuteuguy, 2016-03-14:
Updated code:
var five = require("johnny-five");
var Edison = require("edison-io");
var board = new five.Board({
  io: new Edison()
});

board.on("ready", function() {

  var lcd = new five.LCD({
    controller: "JHD1313M1"
  });

  lcd.useChar("heart");
  lcd.cursor(0, 0).print("I :heart: Johnny-Five");

  var barometer = new five.Barometer({
    controller: 'BMP180'
  });

  barometer.on('data', function() {
    console.log("barometer");
    console.log("  pressure : ", this.pressure);
    console.log("--------------------------------------");
  });

});

And picture: [https://dl.dropboxusercontent.com/u/9822239/IMG_1611.jpeg](https://dl.dropboxusercontent.com/u/9822239/IMG_1611.jpeg)

rwaldron, 2016-03-14:
Thanks, that's helpful. Can you try this program without the sparkfun blocks attached? Just attach the edison directly to the intel arduino breakout, like this: 

[image]

rwaldron, 2016-03-14:
Also, can you list here which blocks you have attached? Thanks again!

teuteuguy, 2016-03-15:
Hi rwaldron,

I've removed the Sparkfun blocks as requested (agree that it could have been a potential problem). 
My setup is same as before, Grove Shield, LCD and Barometer.

I've borrowed another Barometer sensor from friend to test (maybe the one I have is broken).

Exactly same result as before, ie. I2C Could not read 2 Bytes ...

Happy to get on gitter to discuss live with you (assuming timezone is OK).

rwaldron, 2016-03-18:
@teuteuguy I apologize for not replying sooner—I tried to catch you on gitter, but I agree the time zones make that hard! I'm going to try to put together a very basic test case for you to try. 

Also, I really appreciate your patience as we work through this together :)

rwaldron, 2016-03-18:
I think I might have a different sensor here, can you link me to the _exact_ sensor you're using?

rwaldron, 2016-03-18:
Is this it? 

[image]

rwaldron, 2016-03-18:
Adding you image here: 

[image]

rwaldron, 2016-03-18:
Or is it this one? 

[image]

rwaldron, 2016-03-18:
ssh to you Edison and run this: 

```
i2cdump -y 6 0x77
```

And paste the results here

teuteuguy, 2016-03-21:
Hey rwaldron,
any updates on this ?

teuteuguy, 2016-03-21:
Hey rwaldron,
No apologies needed :)

The sensor I have is the latter (ie. the one with Address is 0x77 written on it).
Barometer Sensor (BMP180) v1.0

`i2cdump -y 6 0x77`

`root@edison:~# i2cdump -y 6 0x77
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
10: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
20: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
30: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
40: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
50: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
60: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
70: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
80: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
90: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
a0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
b0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
c0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
d0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
e0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
f0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX`

rwaldron, 2016-03-21:
Well, we're getting somewhere right? :D

Did you try connecting the sensor to other I2C ports on the shield? (I'm guessing you did)

Another update: I've ordered this same sensor for Seeed

teuteuguy, 2016-03-22:
I did try on a different port. Same result :(

rwaldron, 2016-03-22:
Ok, then the best we can do is wait until my sensor arrives and then work together to figure out what's going on—thanks again for your patience :)

teuteuguy, 2016-03-23:
Awesome :)
Let me know if I can help. Thanks

rwaldron, 2016-03-23:
Absolutely—as soon as the sensor arrives, I will let you know