rwaldron/galileo-io #52: Pin name undefined? (Issues regarding a fork)

virgilvox, 2016-08-23:
I get this error in my fork of galileo-io since I added pinModes for the intel joule.

Please see https://github.com/virgilvox/galileo-io 

```
1471987477682 Device(s) unknown
1471987477686 Connected undefined
1471987477691 Repl Initialized
>> Ready!!
/home/root/.node_app_slot/node_modules/johnny-five/lib/board.pins.js:238
    return pin.name === needle || pin.id === needle || pin.port === needle;
              ^

TypeError: Cannot read property 'name' of undefined
    at /home/root/.node_app_slot/node_modules/johnny-five/lib/board.pins.js:238:15
    at Pins.findIndex (native)
    at Function.Pins.identity (/home/root/.node_app_slot/node_modules/johnny-five/lib/board.pins.js:237:23)
    at Pins.(anonymous function) [as isPwm] (/home/root/.node_app_slot/node_modules/johnny-five/lib/board.pins.js:256:40)
    at Led.Controllers.DEFAULT.initialize.value (/home/root/.node_app_slot/node_modules/johnny-five/lib/led/led.js:74:30)
    at new Led (/home/root/.node_app_slot/node_modules/johnny-five/lib/led/led.js:203:10)
    at Board.<anonymous> (/home/root/.node_app_slot/main.js:11:13)
    at emitNone (events.js:72:20)
    at Board.emit (events.js:166:7)
    at nextTickCallbackWith0Args (node.js:420:9)
```

rwaldron, 2016-08-23:
If only I knew what code you ran that produced this result... :P

rwaldron, 2016-08-23:
Looks like your condition that determines platform 13 is stuck inside the condition for platform 2

https://github.com/virgilvox/galileo-io/blob/42bd5502e39c774d4edefaa85b6a6082c6c17320/lib/index.js#L50-L129

virgilvox, 2016-08-23:
I did fix that a few commits ago and still have the issue :/ 

Also used the blink example from galileo-io and changed the pin to 103 which corresponds to an onboard LED

rwaldron, 2016-08-24:
It would be useful to me if whatever code you wanted me to look at was the most recent ;)

My guess is that you don't have an entry for that index. I don't know what else to say ¯\_(ツ)_/¯ if I can't see the whole picture then I can't be any help. My joule came today but I can't get a serial connection.

virgilvox, 2016-08-24:
It was current, its possible the history didnt show right away o.O

I suggest following the windows guide for updating the firmware. Rex said he has yet to update the guide for screen, I was recommended to try minicom on osx. - PM if you need help getting into it. 

Want me to PR what I have if you're gonna hack on it anyway? Saves you the trouble of going through all the pins haha

rwaldron, 2016-08-24:
I've pulled your branch locally and the first thing I noted was that the `pinModes` array had an entry at the zeroth index, but there is no pin 0, which I determined was actually the definition for pin 1. That meant that the whole array was "off by one" and since the definition for pin 103 was not actually at index 103, but "off by one" at index 102, the result was "undefined"

virgilvox, 2016-08-24:
*picardfacepalm.jpg*

Have you given it a go on the board?

virgilvox, 2016-08-24:
That fixed it :)

Blinking the onboard LEDs!

rwaldron, 2016-08-24:
> Have you given it a go on the board?

Yes. I have a branch ready for you right now actually (I had to write tests and what-not)

```
npm install rwaldron/galileo-io#joule
```

It's your branch with my changes, ready for testing

rwaldron, 2016-08-24:
We'll close this when galileo-io#joule lands

rwaldron, 2016-08-24:
Just a heads up: Right now Johnny-Five not working correctly, but you can use galileo-io on its own. I will fix the issue shortly

rwaldron, 2016-08-26:
Apparently, there are alot of regressions in the latest os image.

virgilvox, 2016-08-29:
Such as ?

rwaldron, 2016-08-30:
- https://github.com/intel-iot-devkit/mraa/issues/570 (fixed today)
- And I2C doesn't appear to work at all