rwaldron/galileo-io #55: PingFirmata on the Edison

WoodyWoodsta, 2016-09-17:
I see that no-one appears to have run into this issue wherever I search, so I may as well open an issue here!

I'd like to make use of the `ping` functionality for the `HC-SR04` ultrasonic module on my Intel Edison (which is connected to the Arduino breakout), and I am not able to get the I2C backpack, so I understand that I need to have a custom firmata installed. However, there is no such version available for the Edison as far as I can see.

What exactly is required for there to be support for this functionality? The Johnny-Five site just states "no hardware support" but I would like to know what hardware requirements this needs.

This is a fairly critical part that I need for my thesis and I'm willing to put some time into helping out getting this functionality up and riding, if the hardware allows.

reconbot, 2016-09-17:
You're going to need an i2c backpack for the Edison to use an ultrasonic
sensor that requires precisely timed pulses. Or you'll need a different
sensor that has an analog out or i2c out on it itself. There's a huge list
of parts that will work with the eidson on the proximity api page under
"analog" and "i2c".

http://johnny-five.io/api/proximity/

---
Francis Gulotta
wizard@roborooter.com

WoodyWoodsta, 2016-09-17:
After more reading, I understand that this is also due to Node's event loop and that watching a file change (the GPIO pin) in an "asynchronous" manner is still bound by the event loop reaching the point where it can fire a handler. What a pity, I will have to see what kind of accuracy I can get out of a pure js approach.

I still would not mind knowing why this cannot be rectified in the firmata. Unfortunately I'm not too clued up on how the protocol works!

reconbot, 2016-09-17:
There is no firmata on the Edison. Even with c++ this is difficult or
impossible to use achieve with sysfs. (File based gpio)

WoodyWoodsta, 2016-09-18:
Right! This makes a lot more sense to me now. Thanks for your help.

closed 2016-09-18 by WoodyWoodsta