rwaldron/galileo-io #56: JF + Intel Edison + Grove - Digital Light Sensor

trifroggy, 2016-10-25:
Hello,
I am trying to run this sample:
http://johnny-five.io/examples/grove-light-sensor-edison/

but keep on getting the following error:
ERROR: I2C: Could not read 4 Bytes from peripheral with address 0x39

For sanity check I've ran the basic example with the LCD and it works (same I2C plug, same wire...).
I'm using the baseshield v2 and components are plugged in very much like in the LCD example.

Any suggestions on how to troubleshoot the problem further?
I'll use all the help I can get :)  Thank you!

rwaldron, 2016-10-25:
I'll take a look at this asap. Just a heads up: TSL2561 support is being completely refactored.

trifroggy, 2016-10-25:
No problem. If I can help in any way, let me know!