rwaldron/galileo-io #57: Joule-io - Unknown Exception on "this.name = Pin.IO.getPlatformName();"

virgilvox, 2016-11-04:
So I'm getting this error when running the example for Joule-IO

Any idea why? I'm still running the version of Ostro from back in August. 

```
/home/root/.node_app_slot/node_modules/joule-io/node_modules/galileo-io/lib/galileo.js:221
  this.name = Pin.IO.getPlatformName();
                     ^

Error: Unknown exception
    at Error (native)
    at new Galileo (/home/root/.node_app_slot/node_modules/joule-io/node_modules/galileo-io/lib/galileo.js:221:22)
    at Object.<anonymous> (/home/root/.node_app_slot/main.js:4:7)
    at Module._compile (module.js:409:26)
    at Object.Module._extensions..js (module.js:416:10)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:300:12)
    at Function.Module.runMain (module.js:441:10)
    at startup (node.js:139:18)
    at node.js:968:3
```

virgilvox, 2016-11-04:
Side note: Headed to the first joule hackathon tomorrow

rwaldron, 2016-11-04:
I have no idea 0_o I will try to get this resolved for you today.

Aside: I'm not sure supporting that Ostro build is worth our efforts. @rexstjohn linked me to http://waipio.sakoman.com/feeds/waipio-2.1-ia64/images/current/waipio-image-xfce-intel-corei7-64.hddimg, and using https://etcher.io/ I created an sdcard to boot the joule from.

rwaldron, 2016-11-04:
Are you using the Intel XDK?

rwaldron, 2016-11-04:
Did you freshly `npm install joule-io`? The 0.9.4 version of joule-io was published to claim the name, but support for joule has never actually been released.

rwaldron, 2016-11-04:
To use the development version, you need to install with `npm install rwaldron/galileo-io`

rwaldron, 2016-11-04:
I just want to communicate this: getting this report the day before a hackathon is very distressful. I want the nodebots platform to be massively successful, but I definitely need more advance notice for important events. As far as I knew, there had been no further growth of interest for the Joule, so it hasn't been a priority for two months.

rwaldron, 2016-11-04:
Using the Intel XDK:

- package.json (note that I'm point the dependency at the git repo!!)

  ```
  {
    "name": "xdk-j5",
    "description": "",
    "version": "0.0.0",
    "main": "main.js",
    "engines": {
      "node": ">=0.10.0"
    },
    "dependencies": {
        "johnny-five": "latest",
        "galileo-io": "rwaldron/galileo-io"
    }
  }
  ```
- main.js

  ```js
  var five = require("johnny-five");
  var Joule = require("galileo-io");

  var board = five.Board({
    io: new Joule(),
  });

  board.on("ready", () => {
    console.log("board is ready");

    var led = new five.Led("B2_11");
    led.blink();
  });
  ```

I have a multimeter connected to B2_11 and the GND next to it, as shown here: 

[image]

Here is the running programming output in XDK:

[image]

rwaldron, 2016-11-04:
Another test successful: 

```js
var five = require("johnny-five");
var Joule = require("galileo-io");

var board = five.Board({
  io: new Joule(),
});

board.on("ready", () => {
  console.log("board is ready");

  var leds = new five.Leds(["L0", "L1", "L2", "L3"]);
  leds.blink();
});
```

And all four are blinking in unison.

rwaldron, 2016-11-04:
Successful I2C test: 

```js
var five = require("johnny-five");
var Joule = require("galileo-io");

var board = five.Board({
  io: new Joule(),
});

board.on("ready", () => {
  console.log("board is ready");

  var rgb = new five.Led.RGB({
    controller: "BLINKM"
  });
  var index = 0;
  var rainbow = ["FF0000", "FF7F00", "FFFF00", "00FF00", "0000FF", "4B0082", "8F00FF"];

  board.loop(1000, () => {
    if (index + 1 === rainbow.length) {
      index = 0;
    }
    rgb.color(rainbow[index++]);
  });
});
```

This **REQUIRES** a 3.3kΩ pull-up resistor on both the SDA and SCL lines: 

https://www.youtube.com/watch?v=hl8t1yqBrBs

rwaldron, 2016-11-04:
I've just released {Galileo,Edison,Joule}-IO v0.10.0, so I'm going to test the stuff above again

rwaldron, 2016-11-04:
Ok, updated stuff...

- package.json: 

  ```
  {
    "name": "xdk-j5",
    "description": "",
    "version": "0.0.0",
    "main": "main.js",
    "engines": {
      "node": ">=4.0.0"
    },
    "dependencies": {
      "johnny-five": "latest",
      "joule-io": "^0.10.0"
    }
  }
  ```
- main.js:
  ```
  var five = require("johnny-five");
  var Joule = require("joule-io");

  var board = five.Board({
    io: new Joule(),
  });

  board.on("ready", () => {
    console.log("board is ready");

    var leds = new five.Leds(["L0", "L1", "L2", "L3"]);
    leds.blink();

    var rgb = new five.Led.RGB({
      controller: "BLINKM"
    });
    var index = 0;
    var rainbow = ["FF0000", "FF7F00", "FFFF00", "00FF00", "0000FF", "4B0082", "8F00FF"];

    board.loop(1000, () => {
      if (index + 1 === rainbow.length) {
        index = 0;
      }
      rgb.color(rainbow[index++]);
    });
  });
  ```

virgilvox, 2016-11-04:
You sir are a wizard! <3 Thank you!

I'll test on my end when I get a chance (airport wifi sucks)

I think its important to continue supporting Ostro - since the boards are pre-loaded with that. Loading a new OS can be daunting to a novice. 

I totally dropped the ball on this - feel awful springing it on ya last minute. Kept meaning to pair up to check joule-io was ready. Its not critical to the hackathon, but I intend to preach the good word of J5 as I always do ;)

rwaldron, 2016-11-04:
> I think its important to continue supporting Ostro

All of the above was completed on Ostro; I think what happened is that mraa was missing parts, and those issues have been fixed since August. 

> I totally dropped the ball on this - feel awful springing it on ya last minute. Kept meaning to pair up to check joule-io was ready.

It's ok, I just needed to air it out ;) I'm glad everything came together. You should be testing with joule-io@0.10.0

virgilvox, 2016-11-08:
Good news! Also got this working on Ubuntu LTS during the hackathon. You just gotta make sure to install libmraa and we had to use sudo permissions to run it. 

Going to close this issue :)

closed 2016-11-08 by virgilvox